PelionIoT/mbed-cloud-client-example #4: Porting from mbed-cloud-client-example-restricted

Alex-EEE, 2018-04-24:
Is there a porting guide for moving from the restricted version of this repo to this version?  Or can someone summarize the differences here?  It looks like some things were changed around the pal layer?

teetak01, 2018-04-25:
Hi @Alex-EEE there is no code-diff between public release (this repo) and restricted version (Limited Availability) if the version number is identical. The restricted repository may get from time to time preview to upcoming new features before those are released to public.

Of course if you do find actual code diff between same versions, please let me know.

Alex-EEE, 2018-04-25:
Ok, I see now, I have a version I downloaded and modified from Jan, and there have just been tremendous changes in the repo and structure since then to support the newer version of Cloud Client.  Can you point me at some release notes or just SOME documentation outlining what changes are being made to Cloud Client, why they are made, and how you can update your application to play nice with the new version?

teetak01, 2018-05-02:
There are also CHANGELOGs as part of the repositories:

https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/CHANGELOG.md
https://github.com/ARMmbed/mbed-cloud-client/blob/master/CHANGELOG.md

Official release notes can be found from https://cloud.mbed.com/docs/current/release-notes/index.html

closed 2018-05-02 by Alex-EEE