PelionIoT/mbed-cloud-client-example #9: Build error on Raspbian OS

yemkay, 2018-05-04:
I'm trying to build on Raspbian Stretch OS. I know its not supported, but would still give it a try for setting up my development device. 

Does this error familiar to anyone?

```
(pi@raspberrypi:~/mbed-cloud-client-example $ sudo python pal-platform/pal-platform.py fullbuild --target x86_x64_NativeLinux_mbedtls --toolchain GCC --external ./../define.txt --name mbedCloudClientExample.elf
2018-05-04 13:20:23,869 - pal-platform - INFO - fullBuild running for target = x86_x64_NativeLinux_mbedtls with toolchain = GCC
2018-05-04 13:20:23,870 - pal-platform - INFO - Getting mbedtls from git
2018-05-04 13:20:23,871 - pal-platform - INFO - /home/pi/mbed-cloud-client-example/pal-platform/Middleware/mbedtls/mbedtls already exists, updating from https://github.com/ARMmbed/mbedtls.git
2018-05-04 13:20:30,272 - pal-platform - INFO - Checking out from https://github.com/ARMmbed/mbedtls.git at mbedtls-2.4.2
Deployment for x86_x64_NativeLinux_mbedtls is successful.
Deployment instructions are in /home/pi/mbed-cloud-client-example/pal-platform/x86_x64_NativeLinux_mbedtls.txt.
2018-05-04 13:20:30,317 - pal-platform - INFO - Generated /home/pi/mbed-cloud-client-example/__x86_x64_NativeLinux_mbedtls/autogen.cmake
Generation for x86_x64_NativeLinux_mbedtls is successful, please run cmake & make from /home/pi/mbed-cloud-client-example/__x86_x64_NativeLinux_mbedtls
2018-05-04 13:20:30,319 - pal-platform - INFO - Checking Environment for Toolchain - GCC
2018-05-04 13:20:30,328 - pal-platform - INFO - running cmake
2018-05-04 13:20:30,329 - pal-platform - INFO - setting environment: GCC_DIR = /usr
Traceback (most recent call last):
  File "pal-platform/pal-platform.py", line 1020, in <module>
    cli(sys.argv[1:])
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 1086, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/click/decorators.py", line 64, in new_func
    return ctx.invoke(f, obj, *args[1:], **kwargs)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "pal-platform/pal-platform.py", line 1007, in fullbuild
    runCmakeAndMake(out_dir, isDebug, toolchain, output, envPair, external, name, numOfBuildThreads)  # CMAKE + build debug version
  File "pal-platform/pal-platform.py", line 858, in runCmakeAndMake
    p = subprocess.Popen([command, argument1,argument2, argument3, argument4, argument5], cwd=folder)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
)
```

JaniSuonpera, 2018-05-08:
Hi @yemkay 

What Linux you using for building? There are available supported Linux distros https://www.yoctoproject.org/docs/2.2/ref-manual/ref-manual.html#detailed-supported-distros

Current Rasperry PI Yocto build is using Morty/Yocto 2.2.

JaniSuonpera, 2018-05-08:
Hi @yemkay 

You cannot use Rasperry PI for building mbed-cloud-client example application. It must pass through Yocto build. See https://cloud.mbed.com/docs/v1.2/connecting/tutorial-connect-rpi3.html

closed 2018-05-08 by JaniSuonpera