PelionIoT/mbed-cloud-client-example #10: Some issues with storage on Mbed Cloud 1.3

ccli8, 2018-05-07:
I am evaluating Mbed Cloud 1.3 on **NUMAKER_PFM_M487** target and meet some issues with storage. Most issues are related to `mbed-cloud-client-example/source/platform/mbed-os/common_setup.cpp`:
1. There's `#if (MBED_CONF_STORAGE_SELECTOR_STORAGE  == SD_CARD)` in this file, but I find no `SD_CARD`, `SPI_FLASH`, etc. are defined.
1. It seems this file requires `SD_CARD` to be partitioned (**MBR**). And no-MBR `SD_CARD`, `SPI_FLASH`, etc. are not supported. How to support storage without MBR?
1. When I choose external storage for firmware update candidates, I find these candidates are stored directly in `SD_CARD`, not through filesystem. Clash for filesystem use and raw access in the same SD card?

**Mbed OS**
mbed-os-5.8.3

**mbed-cloud-client-example**
1.3.1.1

**mbed-cloud-client**
1.3.1

teetak01, 2018-05-07:
Hi @ccli8,

1. We also noticed this issue already earlier and it has already been fixed in internal dev. version and will be released in future release.
2. Cloud Client uses KCM for storage and this requires a filesystem currently. We recommend FAT32 with SD-cards, but also LittleFS can be used. It is possible to use SPI_FLASH, where LittleFS is more suitable. Generally any external flash designed for filesystem usage can be used for Cloud Client storage (This is for storing the Client credentials).
3. Update Client uses blockdevice directly, and will allocate part of the SD-card for its own use. This can also use any other external flash as an alternative. I agree that there is potential for conflict with the current configuration (although the data stored there is only a few kilobytes). It would be cleaner to, for. ex. define the first two (or last two) MBs of the SD-card for Update storage, and then partition some space for KCM after that.

Looking at https://os.mbed.com/platforms/NUMAKER-PFM-M487. The internal flash is enough for the application, but for KCM and Update you would need some external flash (SPI_FLASH or SD-card for. ex.). Is there such available?

ccli8, 2018-05-07:
@teetak01 Thanks for your reply. About Issue 2, is storage e.g. SD card required to be partitioned (MBR)? It is very common the whole storage is used as a partition and there's no MBR in it.

ccli8, 2018-05-10:
@teetak01 @JaniSuonpera I partition my SD card manually and it works with the 1.3.1.1 code. Thanks for your information.

closed 2018-05-10 by ccli8