PelionIoT/mbed-cloud-client-example #12: Fail to combine bootloader/application programs of HEX format

ccli8, 2018-05-15:
During my evaluation of Mbed Cloud 1.3.1 on our on-going **NUMAKER_PFM_M2351** target, I fail to combine bootloader/application programs of HEX format. I re-produce the issue on **K64F**:

### Environment
--------------------------------------------------------------------------------------
**Target**
K64F

<pre>
"K64F": {
        "supported_form_factors": ["ARDUINO"],
        "core": "Cortex-M4F",
        "supported_toolchains": ["ARM", "GCC_ARM", "IAR"],
        "extra_labels": ["Freescale", "MCUXpresso_MCUS", "KSDK2_MCUS", "FRDM", "KPSDK_MCUS", "KPSDK_CODE", "MCU_K64F"],
        "is_disk_virtual": true,
        "macros": ["CPU_MK64FN1M0VMD12", "FSL_RTOS_MBED"],
        "inherits": ["Target"],
        "detect_code": ["0240"],
        "device_has": ["ANALOGIN", "ANALOGOUT", "I2C", "I2CSLAVE", "INTERRUPTIN", "LOWPOWERTIMER", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SERIAL_FC", "SERIAL_ASYNCH", "SLEEP", "SPI", "SPI_ASYNCH", "SPISLAVE", "STDIO_MESSAGES", "STORAGE", "TRNG", "FLASH"],
        "features": ["LWIP", "STORAGE"],
        "release_versions": ["2", "5"],
        "device_name": "MK64FN1M0xxx12",
        "bootloader_supported": true,
        <b>
        "OUTPUT_EXT": "hex"
        </b>
    },
</pre>

**mbed-bootloader**
v3.3.0

**mbed-cloud-client-bootloader**
1.3.1.1

**mbed-os**
5.8.3

**ARM Toolchain**
ARM 5.24

**GCC Toolchain**
7-2017-q4-major) 7.2.1

### Steps to reproduce
--------------------------------------------------------------------------------------
To build mbed-bootloader, I run:
`
mbed compile -m K64F -t ARM --profile tiny.json
`
And copy built `mbed-bootloader.hex` to `mbed-cloud-client-example`/`tools` directory.

To build mbed-cloud-client-example, I run:
`
mbed compile -m K64F -t GCC_ARM
`

Then I combine `mbed-bootloader.hex` and `mbed-cloud-client-example` and meet error:
```
$ python tools/combine_bootloader_with_app.py -m K64F -a BUILD/K64F/GCC_ARM/mbed-cloud-client-example_application.hex -o combined.hex
Using bootloader C:\ccli8\iot\mbed\mbed-os\mbed-cloud-client-example\tools\mbed-bootloader-k64f-block_device-sotp-v3_3_0.hex
Loading bootloader from hex file.
imageSize:    352748
imageHash:    0d38da35ae43c35e564185308fd933952ba0849e16c6d0a521a681687084a75e
imageversion: 1526361824
Writing header to address 0x0000a000.
Loading application from hex file
Traceback (most recent call last):
  File "tools/combine_bootloader_with_app.py", line 318, in <module>
    output_fn, args.set_version, args.no_bootloader)
  File "tools/combine_bootloader_with_app.py", line 171, in combine
    ih.fromfile(app_fn, format=app_format)
  File "C:\Python27\lib\site-packages\intelhex\__init__.py", line 246, in loadfile
    self.loadhex(fobj)
  File "C:\Python27\lib\site-packages\intelhex\__init__.py", line 209, in loadhex
    decode(s, line)
  File "C:\Python27\lib\site-packages\intelhex\__init__.py", line 171, in _decode_record
    raise DuplicateStartAddressRecordError(line=line)
intelhex.DuplicateStartAddressRecordError: Start Address Record appears twice at line 22056
```

teetak01, 2018-05-15:
Hi @ccli8, thanks for reporting.

The issue you looks like something that was just fixed in Mbed OS: https://github.com/ARMmbed/mbed-os/pull/6885.

The combine-script can also accept `.bin` format and output `.hex`, thus if you compile your bootloader in .bin format, it should work fine then.

ccli8, 2018-05-15:
@teetak01 Thanks for your information. It would fix my issue.

closed 2018-05-15 by ccli8