PelionIoT/mbed-cloud-client-example #13: Cannot build on Ubuntu

janjongboom, 2018-05-16:
I'm following all the steps in the docs, but building fails: https://cloud.mbed.com/docs/current/connecting/connecting.html

```
-- Generating done
-- Build files have been written to: /home/janjon01/mbed-cloud-client-example/__x86_x64_NativeLinux_mbedtls
Scanning dependencies of target platformCommon
[  1%] Building C object mbed-cloud-client-example/source/platform/CMakeFiles/platformCommon.dir/Linux/common_setup.c.obj
/home/janjon01/mbed-cloud-client-example/source/platform/Linux/common_setup.c:28:17: fatal error: pal.h: No such file or directory
compilation terminated.
mbed-cloud-client-example/source/platform/CMakeFiles/platformCommon.dir/build.make:62: recipe for target 'mbed-cloud-client-example/source/platform/CMakeFiles/platformCommon.dir/Linux/common_setup.c.obj' failed
make[3]: *** [mbed-cloud-client-example/source/platform/CMakeFiles/platformCommon.dir/Linux/common_setup.c.obj] Error 1
CMakeFiles/Makefile2:424: recipe for target 'mbed-cloud-client-example/source/platform/CMakeFiles/platformCommon.dir/all' failed
make[2]: *** [mbed-cloud-client-example/source/platform/CMakeFiles/platformCommon.dir/all] Error 2
CMakeFiles/Makefile2:83: recipe for target 'CMakeFiles/mbedCloudClientExample.elf.dir/rule' failed
make[1]: *** [CMakeFiles/mbedCloudClientExample.elf.dir/rule] Error 2
Makefile:162: recipe for target 'mbedCloudClientExample.elf' failed
make: *** [mbedCloudClientExample.elf] Error 2
2018-05-16 08:32:29,411 - pal-platform - ERROR - ** build failed **
2018-05-16 08:32:29,411 - pal-platform - INFO - 
Completed fullBuild running for target = x86_x64_NativeLinux_mbedtls
With toolchain = GCC.
Output directory: /home/janjon01/mbed-cloud-client-example/out
```

janjongboom, 2018-05-17:
@teetak01 How would I deploy this? It's not in the instructions...

teetak01, 2018-05-17:
Instructions suggest to use` mbed import`, which performs `mbed deploy` automatically.

Alternatively, if you clone the application via `git clone,` then you need to run `mbed deploy` seperately.

janjongboom, 2018-05-17:
@teetak01 Interesting, I missed the mbed import part. but this is also pulling in mbed OS and all the dependencies like storage-selector. I don't like that.

teetak01, 2018-05-18:
I do not see that as a real problem. Pure linux-application does not include those libs at all. This reference application supports both Linux and Mbed OS out-of-box so some compromise needs to be done. We really would not want to write yet-another tool to deploy components seperately for Linux.

teetak01, 2018-05-21:
Closing this issue as resolved.

closed 2018-05-21 by teetak01