PelionIoT/mbed-cloud-client-example #14: can't build on CLI for target nRF_52840

surajdagar, 2018-05-18:
Error Logs:

Compile [ 37.9%]: GenericGap.cpp
[Warning] BLE.h@1049,39: 'virtual ble_error_t Gap::disconnect(Gap::DisconnectionReason_t)' is deprecated: Use disconnect(Handle_t, DisconnectionReason_t) instead. [-Wdeprecated-declarations]
Compile [ 38.1%]: GenericSecurityManager.cpp
[Warning] BLE.h@1049,39: 'virtual ble_error_t Gap::disconnect(Gap::DisconnectionReason_t)' is deprecated: Use disconnect(Handle_t, DisconnectionReason_t) instead. [-Wdeprecated-declarations]
[Error] GenericSecurityManager.h@357,15: 'public_key_t' does not name a type; did you mean 'public_key_coord_t'?
[Error] GenericSecurityManager.h@358,15: 'public_key_t' does not name a type; did you mean 'public_key_coord_t'?
[Error] GenericSecurityManager.cpp@722,6: no matching function for call to 'ble::generic::GenericSecurityManager::crypto_toolbox_f4(const public_key_coord_t&, const public_key_coord_t&, ble::oob_lesc_value_t&, ble::oob_confirm_t&)'
[Error] GenericSecurityManager.cpp@756,6: prototype for 'bool ble::generic::GenericSecurityManager::crypto_toolbox_f4(const public_key_coord_t&, const public_key_coord_t&, const oob_lesc_value_t&, ble::oob_confirm_t&)' does not match any in class 'ble::generic::GenericSecurityManager'
[Error] GenericSecurityManager.h@356,17: candidate is: static bool ble::generic::GenericSecurityManager::crypto_toolbox_f4(const int&, const int&, const oob_lesc_value_t&, ble::oob_confirm_t&)
[Error] GenericSecurityManager.cpp@1049,5: no matching function for call to 'ble::generic::GenericSecurityManager::crypto_toolbox_f4(const public_key_coord_t&, const public_key_coord_t&, ble::oob_lesc_value_t&, ble::oob_confirm_t&)'
[ERROR] In file included from ./mbed-os/features/FEATURE_BLE/ble/generic/GenericGap.h:22:0,
                 from ./mbed-os/features/FEATURE_BLE/ble/generic/GenericSecurityManager.h:25,
                 from .\mbed-os\features\FEATURE_BLE\source\generic\GenericSecurityManager.cpp:19:
./mbed-os/features/FEATURE_BLE/ble/BLE.h: In member function 'ble_error_t BLE::disconnect(Gap::DisconnectionReason_t)':
./mbed-os/features/FEATURE_BLE/ble/BLE.h:1049:39: warning: 'virtual ble_error_t Gap::disconnect(Gap::DisconnectionReason_t)' is deprecated: Use disconnect(Handle_t, DisconnectionReason_t) instead. [-Wdeprecated-declarations]
         return gap().disconnect(reason);
                                       ^
I

yogpan01, 2018-05-21:
Hi @surajdagar ,
This is mbed-os build issue where you would need to have the correct setup for this board.
@dlfryar Do you have any pointers for compiling client on this board ?

dlfryar, 2018-05-21:
Latest build steps/script and config attached.

[build.sh.txt](https://github.com/ARMmbed/mbed-cloud-client-example/files/2024232/build.sh.txt)
[mbed_lib.json.txt](https://github.com/ARMmbed/mbed-cloud-client-example/files/2024233/mbed_lib.json.txt)

teetak01, 2018-05-29:
@surajdagar is this issue resolved for you?

closed 2018-06-01 by surajdagar