PelionIoT/mbed-cloud-client-example #28: Problem with importing project to IDE

radzioon, 2018-11-15:
Hi,

I want to modify your example but I have a problem with importing the project to IDE (CLion). The problem is that cmake don't see common.cmake file and unrecognise expressions like 
`SET (CMAKE_CXX_COMPILER "/usr/bin/clang++")`

Maybe someone is able to write some instruction on how to start with it?

yogpan01, 2018-12-04:
Hi,
The CMake setup has been tested in command line and a specific IDE (CLion) setup would require configuration changes from your side. The CMake setup here is a reference for starting up project over command line. We will however try to check if we can provide any further assistance for this issue but I would suggest you to refer documentation for your IDE to import custom project and configure their toolchain setting.

teetak01, 2019-06-20:
closing as there has been no response for several months.

closed 2019-06-20 by teetak01