PelionIoT/mbed-cloud-client-example #29: How to build the LITE configuration ?

kaidokert, 2018-11-18:
I'm not sure if the public repo here is supposed to build with ARM_UC_PROFILE_MBED_CLIENT_LITE set ? I've tried both 1.5 and 2.0 releases and it seems that the codepath for lite client couldn't even be compiled because it's using some outdated APIs. 

I made a test repo here that builds cloud-client on Travis for both 1.5 and 2.0 versions

https://github.com/iconservo/mbed-testbuild/

It builds K64F, K64F-Wifi, Odin and F411 just fine and publishes [binary build releases](https://github.com/iconservo/mbed-testbuild/releases). For a good measure i added in two different armgcc compiler versions as well

Trying to [build Lite branch](https://github.com/iconservo/mbed-testbuild/tree/test-lite-build) here isn't going well though. I added `ARM_UC_PROFILE_MBED_CLIENT_LITE=1` in the config, but the build fails:
https://travis-ci.com/iconservo/mbed-testbuild/jobs/159359486

```
[Warning] arm_uc_pre_shared_key.c@66,35: implicit declaration of function 'get_config_parameter' [-Wimplicit-function-declaration]
[Error] arm_uc_pre_shared_key.c@66,56: 'UPDATE_PSK_SECRET' undeclared (first use in this function)
[ERROR] ./mbed-cloud-client/update-client-hub/modules/control-center/source/arm_uc_pre_shared_key.c: In function 'ARM_UC_PreSharedKey_GetSecret':
./mbed-cloud-client/update-client-hub/modules/control-center/source/arm_uc_pre_shared_key.c:66:35: warning: implicit declaration of function 'get_config_parameter' [-Wimplicit-function-declaration]
         ccs_status_e ccs_status = get_config_parameter(UPDATE_PSK_SECRET,
                                   ^~~~~~~~~~~~~~~~~~~~
./mbed-cloud-client/update-client-hub/modules/control-center/source/arm_uc_pre_shared_key.c:66:56: error: 'UPDATE_PSK_SECRET' undeclared (first use in this function)
         ccs_status_e ccs_status = get_config_parameter(UPDATE_PSK_SECRET,
                                                        ^~~~~~~~~~~~~~~~~
./mbed-cloud-client/update-client-hub/modules/control-center/source/arm_uc_pre_shared_key.c:66:56: note: each undeclared identifier is reported only once for each function it appears in
```

Looks like `get_config_parameter` and a few other related APIs have been removed a while ago. Trying to stub them out gets messy real fast, too.  Am i going about this wrong ?

teetak01, 2018-11-19:
Hi @kaidokert, unfortunately what you are trying is not possible. Client Lite is a different product.

This repository is for Pelion Device Management Client example (old Mbed Cloud Client).

The repository (restricted access) for Pelion Device Management Client Lite example (old Mbed Client Lite) is https://github.com/ARMmbed/mbed-client-lite-example-restricted.

The products do share some common code (mainly Update Client), and have common historical root, but are not currently compatible and cannot be treated as different modes of the Client.

JanneKiiskila, 2018-11-19:
Hi @kaidokert - Teemu is indeed correct. They are two separate products and Lite client has it's own repositories([example](https://github.com/ARMmbed/mbed-client-lite-example-restricted.) and [library](https://github.com/ARMmbed/mbed-client-lite-restricted.)) . You should have access to it soonest. Please ensure the [security guidelines for Arm GitHub users](https://os.mbed.com/blog/entry/Security-update-for-all-users-of-the-ARM/) are set in in your GitHub account.

Documentation for Client Lite is available here - https://cloud.mbed.com/docs/current/connecting/device-management-client-lite.html.

I will close this ticket, as there is no compilation issue.

closed 2018-11-19 by JanneKiiskila