PelionIoT/mbed-cloud-client-example #41: mbed-cloud-client example run in ARM9 system problem

mjchen1, 2019-03-21:
1. I use "python pal-platform/pal-platform.py fullbuild --target ARM_OpenWRT_mbedtls --toolchain GCC-OPENWRT --external ./../define.txt --name mbedCloudClientExample.elf" can build success.
but when I run ".elf" in my system will occur error.
2. when I run "/out/Release/mbedCloudClientExample.elf" will occur "MBEDTLS_ERR_ECP_INVALID_KEY".
this error come from the function "static int ecp_check_pubkey_sw( const mbedtls_ecp_group *grp, const mbedtls_ecp_point *pt )".
3. but, when I run "/out/Debug/mbedCloudClientExample.elf" will occur "MBEDTLS_ERR_X509_INVALID_FORMAT".
This error come from the function "static int x509_crt_parse_der_core( mbedtls_x509_crt *crt, const unsigned char *buf, size_t buflen )"
4. my system cpu is "arm926ej-s" and I run in Linux_4.4 Kernel.
5. why run Release/.elf and Debug/.elf have different result..??
6. and, how can I solve these error...??(any suggestion..??)

JanneKiiskila, 2019-06-03:
Any updates on this @mjchen1 ? It's quite hard to offer assistance on this end, as we don't know the full setup (board, compiler versions etc.). Arm9 systems in itself should not be a problem, it definitely should work - the code is just C/C++.

teetak01, 2019-08-16:
closing due inactivity.

closed 2019-08-16 by teetak01