PelionIoT/mbed-cloud-client-example #45: Request: Add support for for CY8CPROTO-062-4343W

loverdeg-ep, 2019-06-07:
Very low cost board with high end capability.

We want to prototype for another product with this kit.

https://www.cypress.com/documentation/development-kitsboards/psoc-6-wi-fi-bt-prototyping-kit-cy8cproto-062-4343w

@maclobdell 
@farrenv

yogpan01, 2019-06-10:
@JanneKiiskila can you please check this one?

JanneKiiskila, 2019-06-10:
We already support one version of PSOC6, apparently they use the same board support, so it might work out of the box or with minor modifications. Could you give it a try  @loverdeg-ep ?

loverdeg-ep, 2019-06-10:
@farrenv Lets test this when you are back in office.

JanneKiiskila, 2019-06-11:
Example configuration for the PSOC6 Pioneer Kit available here; https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/configs-psa/wifi_esp8266_v4.json

JanneKiiskila, 2019-06-17:
Please keep us in the loop & give us feedback, if you have any issues.

maclobdell, 2019-06-24:
I'm trying to get this example (version 3.1.1) compiling for CY8CPROTO-062-4343W.  

I updated to mbed-os 5.12.4.

Config file tested is the following.

```
        "CY8CPROTO_062_4343W": {
            "target.features_remove"                       : ["BLE"],
            "target.extra_labels_remove"               : [ "CORDIO" ],			
  		    "target.macros_add"                            : ["MBED_ROM_START=0x10002000", 
                                                              "MBED_ROM_SIZE=0x1F6000"
                                                             ],
            "client_app.mbedtls-user-config-file"       : "\"mbedTLSConfig_mbedOS.h\"",
            "client_app.pal-user-defined-configuration" : "\"pal_config_MbedOS.h\"",															 
            "target.network-default-interface-type"     : "WIFI",
            "target.bootloader_img"                     : "tools/mbed-bootloader-cy8cproto_062_4343W.hex",
            "target.header_offset"                      : "0xa000",
            "target.app_offset"                         : "0xa400",
            "update-client.bootloader-details"          : "0x00007188",
            "update-client.application-details"            : "(MBED_ROM_START + 40*1024)",
            "update-client.storage-address"             : "(1024*1024*64)",
            "update-client.storage-size"                : "((MBED_ROM_START + MBED_ROM_SIZE - 40*1024) * MBED_CONF_UPDATE_CLIENT_STORAGE_LOCATIONS)",
            "mbed-cloud-client.update-storage"          : "ARM_UCP_FLASHIAP_BLOCKDEVICE",
            "storage_filesystem.internal_base_address"     : "0x100F8000",
            "storage_filesystem.rbp_internal_size"         : "(2*4*1024)",
            "storage_filesystem.external_base_address"  : "(0x0)",
            "storage_filesystem.external_size"          : "(1024*1024*64)",
            "storage.storage_type"                      : "FILESYSTEM",
            "storage_filesystem.filesystem"             : "LITTLE",
            "storage_filesystem.blockdevice"            : "QSPIF"

```

Initially I get a linker fails due to conflicting BLE functions.  I removed BLE.  Will investigate that.  

Then I get the following linker error.  

```
nk: mbed-cloud-client-example
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Error] @0,0: L6218E: Undefined symbol pal_plat_osEntropyInject (referred from BUILD/CY8CPROTO_062_4343W/ARM/mbed-cloud-client/mbed-client-pal/Source/PAL-Impl/Modules/Entropy/pal_entropy.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol pal_plat_osEntropyInject (referred from BUILD/CY8CPROTO_062_4343W/ARM/mbed-cloud-client/mbed-client-pal/Source/PAL-Impl/Modules/Entropy/pal_entropy.o).
Finished: 0 information, 1 warning and 1 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol pal_plat_osEntropyInject (referred from BUILD/CY8CPROTO_062_4343W/ARM/mbed-cloud-client/mbed-client-pal/Source/PAL-Impl/Modules/Entropy/pal_entropy.o).
Finished: 0 information, 1 warning and 1 error messages.
```

maclobdell, 2019-06-24:
@JanneKiiskila - please let us know if you have any suggestions on this problem.

`Undefined symbol pal_plat_osEntropyInject`

JanneKiiskila, 2019-06-24:
@evgenibo - can you give a hand here? Seems the entropy injection is not found. @maclobdell  - do you have PSA on?

JanneKiiskila, 2019-06-24:
I think the lack of PSA might explain this, please try adding:

```
            "target.extra_labels_add"                    : [ "PSA" ],
```

maclobdell, 2019-06-24:
I was trying to test without PSA.  This target doesn't have PSA configurations in targets.json for mbed-os.

loverdeg-ep, 2019-06-24:
@trowbridgec

evgenibo, 2019-06-24:
I was able to compile it with lastest mbed-cloud-client-example-internal:
https://github.com/ARMmbed/mbed-cloud-client-example-internal. It comes with 5.13.0 rc3

loverdeg-ep, 2019-06-24:
Be nice if I could access the internal example...

evgenibo, 2019-06-24:
I still couldn't reproduce the linker error when using release 3.1.1: https://github.com/ARMmbed/mbed-cloud-client-example/tree/3.1.1 and the config file above.
Can you please add detailed steps including the build command?

JanneKiiskila, 2019-06-24:
Seems the bootloader support at least is missing from `targets.json`, issue raised to mbed-os: https://github.com/ARMmbed/mbed-os/issues/10892

evgenibo, 2019-06-25:
@loverdeg-ep , @maclobdell ,
you need to add two configurations :
1. ```"target.extra_labels_add"  : [ "PSA" ]``` as Janne mentioned
2. add ```MBEDTLS_PSA_CRYPTO_C``` macro.
Those required because PSA API for entropy injection is used, even if the board does not support/configured to work with PSA. No other PSA APIs are used.

JanneKiiskila, 2019-06-25:
Please note that the bootloader is apparently big, 
```
            "mbed-bootloader.bootloader-size"              : "(40*1024)",
```
so the default [`mbed_app.json`](https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/mbed_app.json#L3) in the root of the example cant' be used as is, as it is defaulting to 32 kB bootloader. The `.json` files under [`config-psa`](https://github.com/ARMmbed/mbed-cloud-client-example/tree/master/configs-psa) would work better as a starting point.

maclobdell, 2019-06-26:
I found out a few more things today, but I still can't get this working.  @evgenibo - please let me know if this sounds correct.  

First, since there isn't a CMSIS pack for this target, there are a few defines that have to be made manually.  I believe these should be made in targets.json for this target.

```
        "bootloader_supported" : true,
        "mbed_rom_start" : "0x10002000",
        "mbed_rom_size" : "0x1F6000",
        "sectors": [[268435456, 512]]
```

The Cortex M0+ has a binary application that gets placed at 0x10000000 to 0x10002000.  Then Cortex M4 code goes at 0x10002000.  First with the Pelion bootloader, then application header, then application.  

I tried placing the KVStore at 0x1000D000, the header at 0x1000F000, and app at 0x1000F400.

I tried this configuration of the bootloader.  
```
        "CY8CPROTO_062_4343W": {
            "target.features_remove"                       : [ "BLE" ],
            "storage_filesystem.internal_base_address"     : "0x1000D000",
            "storage_filesystem.rbp_internal_size"         : "(2*4*1024)",
            "update-client.application-details"            : "(MBED_ROM_START + 44*1024)",
            "mbed-bootloader.default-max-application-size" : "(MBED_ROM_START + MBED_ROM_SIZE - MBED_CONF_MBED_BOOTLOADER_APPLICATION_START_ADDRESS - MBED_CONF_STORAGE_FILESYSTEM_RBP_INTERNAL_SIZE)",
            "mbed-bootloader.bootloader-size"              : "(44*1024)",
            "platform.stdio-baud-rate"                     : 115200,
            "target.hex_filename"                          : null,
            "target.device_has_remove"                     : [
                                                              "SERIAL_ASYNCH",
                                                              "SERIAL_FC",
                                                              "PORTOUT",
                                                              "PORTINOUT",
                                                              "RTC",
                                                              "I2C_ASYNCH",
                                                              "SPI_ASYNCH",
                                                              "STDIO_MESSAGES",
                                                              "LPTICKER",
                                                              "TRNG",
                                                              "CRC",
                                                              "PORTIN",
                                                              "PWMOUT",
                                                              "ANALOGIN",
                                                              "ANALOGOUT",
                                                              "I2C"
                                                             ]
        },		
```

And I tried this configuration in the application.
```
        "CY8CPROTO_062_4343W": {
            "target.features_remove"                       : ["BLE"],
            "target.extra_labels_remove"               : [ "CORDIO" ],				
            "client_app.mbedtls-user-config-file"       : "\"mbedTLSConfig_mbedOS.h\"",
            "client_app.pal-user-defined-configuration" : "\"pal_config_MbedOS.h\"",			
            "target.extra_labels_add"                   : [ "PSA" ],
  		    "target.bootloader_supported"               : true,
            "target.network-default-interface-type"     : "WIFI",
            "target.bootloader_img"                     : "tools/mbed-bootloader-cy8cproto_062_4343W.hex",
            "target.header_offset"                      : "0xD000",
            "target.app_offset"                         : "0xD400",
            "update-client.bootloader-details"          : "0x10009188",
            "update-client.application-details"            : "(MBED_ROM_START + 44*1024)",
            "update-client.storage-address"             : "(1024*1024*64)",
            "update-client.storage-size"                : "((MBED_ROM_START + MBED_ROM_SIZE - 48*1024) * MBED_CONF_UPDATE_CLIENT_STORAGE_LOCATIONS)",
            "mbed-cloud-client.update-storage"          : "ARM_UCP_FLASHIAP_BLOCKDEVICE",
            "storage_filesystem.internal_base_address"     : "0x1000D000",
            "storage_filesystem.rbp_internal_size"         : "(2*4*1024)",
            "storage_filesystem.external_base_address"  : "(0x0)",
            "storage_filesystem.external_size"          : "(1024*1024*64)",
            "storage.storage_type"                      : "FILESYSTEM",
            "storage_filesystem.filesystem"             : "LITTLE",
            "storage_filesystem.blockdevice"            : "QSPIF"
        },				
```

However, when I flash and run it, I get this.  
```
[WARN] Couldn't find Erase Type for 4KB size
[BOOT] Mbed Bootloader
[BOOT] ARM: 0000000000000000000000000000000000000000
[BOOT] OEM: 0000000000000000000000000000000000000000
[BOOT] Layout: 0 1000B398
[BOOT] Active firmware integrity check:
[ERR ] Active firmware integrity check failed
[ERR ] Couldn't find valid DeviceKey - err: -2130771705
[BOOT] Slot 0 is empty
[ERR ] Active firmware invalid
[ERR ] Failed to jump to application!
```

Can anyone spot an error in the configurations?

maclobdell, 2019-06-26:
Making progress.  Now able to get application to boot.  

In the bootloader and application configs, I changed to this.

```
           "update-client.application-details"            : "0x1000F000",
```

Now it is able to boot the application.  But it gets stuck making the WiFi connection.

```
[WARN] Couldn't find Erase Type for 4KB size
[BOOT] Mbed Bootloader
[BOOT] ARM: 0000000000000000000000000000000000000000
[BOOT] OEM: 0000000000000000000000000000000000000000
[BOOT] Layout: 0 1000B398
[BOOT] Active firmware integrity check:
[BOOT] [++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++]
[BOOT] SHA256: EA5E9B3001BCDB191F48B78C560E1D9D8FEB88F7BF50CE6D2502578D9A53F436
[BOOT] Version: 1561517193
[ERR ] Couldn't find valid DeviceKey - err: -2130771705
[BOOT] Slot 0 is empty
[BOOT] Active firmware up-to-date
[BOOT] Application's start address: 0x1000F400
[BOOT] AppløApplication ready. Build at: Jun 25 2019 21:46:32
Mbed OS version 5.13.0
mcc_platform_init_connection()
```

maclobdell, 2019-09-11:
Update:  We have a good recipe to demonstrate it working, however there were a few hacks to Mbed OS required in 5.12.x and 5.13.x.  We were waiting on fixes in Mbed OS 5.14 to get bootloader mode working, but there is still an outstanding issue.  Hopefully soon we'll be able to make a PR.

teetak01, 2019-12-11:
There is a PR in https://github.com/ARMmbed/mbed-os-example-pelion/pulls which should be updated to add this support.

closed 2019-12-11 by teetak01