PelionIoT/mbed-cloud-client-example #47: Build error for LPC55S69_NS target

toyowata, 2019-06-13:
# Version

mbed-cloud-client-example (#6eb5211a1d14, tag: 3.2.0)
|- drivers/wifi-x-nucleo-idw01m1 (#a00282000178, tag: v1.2.0)
|- mbed-cloud-client (#4f4a8463dd06, tag: 3.2.0)
`- mbed-os (#73f096399b4c, tags: latest, mbed-os-5.12.4)

# Step to reproduce

```
$ mbed import https://github.com/ARMmbed/mbed-cloud-client-example
$ cd mbed-cloud-client-example
$ mbed dm init -d "arm.com" --model-name "test" -q --force
$ mbed compile -m LPC55S69_NS -t arm --app-config configs-psa/wifi_esp8266_v4.json 
```

# Build log
```
$ mbed compile -m LPC55S69_NS -t arm --app-config configs-psa/wifi_esp8266_v4.json 
[mbed] Working path "/Users/toywat01/test/nxp/mbed-cloud-client-example" (program)
[mbed] WARNING: Could not find mbed program in current path "/Users/toywat01/test/nxp/mbed-cloud-client-example".
       You can fix this by calling "mbed new ." in the root of your program.
---
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: pywin32
       You can install all missing modules by running "pip install -r requirements.txt" in "/Users/toywat01/test/nxp/mbed-cloud-client-example/mbed-os"
       On Mac you might have to install packages as your user by adding the "--user" flag
---
Building project mbed-cloud-client-example (LPC55S69_NS, ARMC6)
Scan: mbed-cloud-client-example
Compile [  0.1%]: SPWFSA04.cpp

(snip)

Compile [ 14.5%]: arm_uc_pal_delta_paal_original_reader.c
[Error] arm_uc_pal_delta_paal_original_reader.c@67,25: use of undeclared identifier 'APPLICATION_ADDR'
[ERROR] ./mbed-cloud-client/update-client-hub/modules/delta-paal/source/arm_uc_pal_delta_paal_original_reader.c:67:25: error: use of undeclared identifier 'APPLICATION_ADDR'
    uint32_t appStart = MBED_CONF_APP_APPLICATION_START_ADDRESS;
                        ^
./mbed-cloud-client/update-client-hub/modules/delta-paal/source/arm_uc_pal_delta_paal_original_reader.c:29:49: note: expanded from macro 'MBED_CONF_APP_APPLICATION_START_ADDRESS'
#define MBED_CONF_APP_APPLICATION_START_ADDRESS APPLICATION_ADDR
                                                ^
1 error generated.

[mbed] ERROR: "/Users/toywat01/mbed-venv/bin/python" returned error.
       Code: 1
       Path: "/Users/toywat01/test/nxp/mbed-cloud-client-example"
       Command: "/Users/toywat01/mbed-venv/bin/python -u /Users/toywat01/test/nxp/mbed-cloud-client-example/mbed-os/tools/make.py -t arm -m LPC55S69_NS --source . --build ./BUILD/LPC55S69_NS/ARM --app-config configs-psa/wifi_esp8266_v4.json"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

teetak01, 2019-06-13:
Hi @toyowata thanks for the report.

If you look at our tutorial https://www.pelion.com/docs/device-management/current/connecting/mbed-os.html, we mention this:

**Note: The --app-config option is not supported, as the application configuration files contain relative paths to the bootloader files.**

If you instead copy the config file to the application root, then it should compile just fine.

```
cp configs-psa/wifi_esp8266_v4.json .
mbed compile -m LPC55S69_NS -t arm --app-config wifi_esp8266_v4.json --profile release
```
Also only release profile is supported due to the limited flash-size in combination with PSA.

https://www.pelion.com/docs/device-management/current/release-notes/device-management-client.html

**NXP LPC55S69:
The board has only 640 KB flash. PSA takes 192 KB out of it.
You can use the Client example (with firmware update and bootloader) with release profile due to the flash size limitation.
Only ARMC6 is supported for compilation.**

Considering the above limitations, this target compiles for me.

toyowata, 2019-06-14:
@teetak01 Thanks.  I can compile the project without errors.
I got runtime error on the LPC55S69 and will raise other issue.
I close this issue.

closed 2019-06-14 by toyowata