PelionIoT/mbed-cloud-client-example #52: Request to update the MbedTLS to 2.19

vince-zeng, 2019-09-16:
When compile the mbed-cloud-client-example for ARM9 platform, we meet the following error. And it should be fixed in  Mbed TLS 2.19 and Mbed Crypto 2.0.0. Would you please update them?
```
make[5]: Entering directory '/home/yangweipeng/only_complie/r3000_3.0/3.0/build_dir/target-r3000/mbed-cloud-client-example/__ARM_Ros_mbedtls'
[ 3%] Building C object mbedtls/crypto/library/CMakeFiles/mbedcrypto.dir/aes.c.obj
[ 5%] Building C object mbedtls/crypto/library/CMakeFiles/mbedcrypto.dir/aesni.c.obj
[ 5%] Building C object mbedtls/crypto/library/CMakeFiles/mbedcrypto.dir/arc4.c.obj
[ 5%] Building C object mbedtls/crypto/library/CMakeFiles/mbedcrypto.dir/aria.c.obj
[ 5%] Building C object mbedtls/crypto/library/CMakeFiles/mbedcrypto.dir/asn1parse.c.obj
[ 5%] Building C object mbedtls/crypto/library/CMakeFiles/mbedcrypto.dir/asn1write.c.obj
[ 6%] Building C object mbedtls/crypto/library/CMakeFiles/mbedcrypto.dir/base64.c.obj
[ 6%] Building C object mbedtls/crypto/library/CMakeFiles/mbedcrypto.dir/bignum.c.obj
/tmp/cc9TmODR.s: Assembler messages:
/tmp/cc9TmODR.s:80: Error: selected processor does not support ARM mode `umaal r1,lr,r3,r0'
/tmp/cc9TmODR.s:84: Error: selected processor does not support ARM mode `umaal r1,lr,r3,r0'
/tmp/cc9TmODR.s:88: Error: selected processor does not support ARM mode `umaal r1,lr,r3,r0'
/tmp/cc9TmODR.s:92: Error: selected processor does not support ARM mode `umaal r1,lr,r3,r0'
/tmp/cc9TmODR.s:96: Error: selected processor does not support ARM mode `umaal r1,lr,r3,r0'
```

teetak01, 2019-09-16:
This is already planned to be part of next release.

teetak01, 2019-09-25:
Mbed TLS was updated to 2.19.1 in #53

closed 2019-09-25 by teetak01