PelionIoT/mbed-cloud-client-example #59: What combinations of wisun domain, class, and mode are supported?

dannywahlquist, 2020-01-03:
I was particularly trying to run PHxA class 2, mode 3 to have only 7 channels.

            "mbed-mesh-api.wisun-regulatory-domain"     : "10",
            "mbed-mesh-api.wisun-operating-class"       : "2",
            "mbed-mesh-api.wisun-operating-mode"        : "3",

yogpan01, 2020-01-06:
@mikter Can you please provide details for these configurations?

dannywahlquist, 2020-01-06:
WiSUN Philippines domain 0X0A, class 2, mode 3 supports Channel Spacing (400kHz), Total Number of Channels 7, Channel Center Freq0 (915.4 MHz).

mikter, 2020-01-07:
We currently only support EU, North America, Korea regulatory domains. as we have not included the conversion table for all regulatory domains, modes and class from the Wi-SUN PHY specification.

If you must use the Philippines Regulatory domain it can be added in this file 
https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_common.c
to ws_common_regulatory_domain_config() function.

dannywahlquist, 2020-01-07:
Worked perfectly, thanks!

closed 2020-01-07 by dannywahlquist