PelionIoT/mbed-cloud-client-example #60: Wi-SUN TLS cert check fail - Pelion

eriknayan, 2020-01-10:
### Description of defect

Following the Wi-SUN tutorial below, we had an issue with the EAP-TLS handshake, which is probably caused by a certification check problem. Link:
https://www.pelion.com/docs/device-management/current/connecting/wi-sun-tutorial.html

#### Target(s) affected by this defect ?
- NXP K64F with I-NUCLEO-S2868A1 RF shield 
- NXP K64F with I-NUCLEO-S2868A1 RF shield  + QUECTEL EC25-AU

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM - gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed CLI 1.10.2

#### How is this defect reproduced ?
The issue is detected after following the tutorial procedure. The border router board seems to be working and acquiring its IPV6 address. We also tested with the minimal mesh example project and it worked without any issues. We followed the procedure to erase the SD card as it had another certificate on it. 

Thanks for your help, attached are the log files for the nanostack-border-router and mbed-cloud-client-example.

[mbed-cloud-client-example-log.txt](https://github.com/ARMmbed/mbed-cloud-client-example/files/4047424/mbed-cloud-client-example-log.txt)
[nanostack-border-router-log.txt](https://github.com/ARMmbed/mbed-cloud-client-example/files/4047425/nanostack-border-router-log.txt)

yogpan01, 2020-01-11:
@mikter Can you please check the logs and provide some input on what can be the potential solution for this issue?

mikaleppanen, 2020-01-13:
Are you using default Wi-SUN certificates provided in application? Problem could be e.g. due to TLS configuration. @eriknayan could you check the configuration you are using for cloud client? E.g. not having this line: 

`
            "client_app.mbedtls-user-config-file"       : "\"configs/wisun_mbedTLSConfig_mbedOS.h\"",
`

on K64F configuration could result error like this since for WI-SUN default certificates, the PEM parsing needs to be enabled.

eriknayan, 2020-01-13:
Hello @mikaleppanen,

First, thanks for you quick answer and help. Indeed, we were missing that line in our .JSON config file for the K64F board. Also, we had changed the Wi-SUN certificates, so after coming back to the default ones it worked as expected.

Could you please clarify how to generate our own Wi-SUN certificates and what's the purpose of each one? We didn't find any doc about this subject.

Thank you

mikaleppanen, 2020-01-14:
@eriknayan Instructions/tools how to acquire certificates are provided by the Wi-SUN alliance.

closed 2020-01-17 by eriknayan