PelionIoT/mbed-cloud-client-example #63: Wi-SUN Node connection with cloud error.

niravdesai712, 2020-02-28:
I am trying to connect my WiSUN node with Pelion cloud. Node successfully connect with Border Router. But node is not able to connect to cloud. Log of Node attached. 
[NodeLog.txt](https://github.com/ARMmbed/mbed-cloud-client-example/files/4266174/NodeLog.txt)

Thanks

teetak01, 2020-02-28:
Hi @niravdesai712 

The issue seems to be related to this:
```
Error occurred : MbedCloudClient::ConnectDnsResolvingFailed
Error code : 12
Error details : Client in reconnection mode DnsResolvingFailed
```
And it looks like timeout from DNS API.

Which version of Client and Mbed OS are you using?

In 4.1.0 version of the example we added these lines for the WiSUN:
https://github.com/ARMmbed/mbed-cloud-client-example/blame/master/configs/mesh_wisun.json#L36-L38

the Mbed OS default timeouts are too short for WiSUN, thus that was fixed in application configuration.

If you already use the latest, then I would verify that your network has access to DNS.

niravdesai712, 2020-02-28:
[App_Config_json.txt](https://github.com/ARMmbed/mbed-cloud-client-example/files/4266244/App_Config_json.txt)
This is my json file for node . I am using NUCLEO-F429ZI board as Node

teetak01, 2020-02-28:
Looks ok. I presume you have very few nodes in the network?

I would then check if the DNS works on that network. For. example if you could just run Client on ethernet on the same network, instead of BR + Node.

niravdesai712, 2020-02-28:
yes i checked with only client without wisun configuration and i get success to cloud . then i start with wisun node and get failure . in my network only one node and one BR .

teetak01, 2020-02-28:
Can you check if you have IPv6 connectivity in your network? It seems that you are getting link-local address from Wi-SUN. fd00:7283:7e00:0:90af:a769:ada8:9f3c

juhhei01, 2020-02-28:
It is actually ULA which is only used inside wi-sun for back-up purpose from create a MESH network ready. Currently Border router generate that if it not learn Global prefix from Backhaul connection.

niravdesai712, 2020-02-28:
so what changes require on  Border router side?

juhhei01, 2020-02-28:
Do you have Native IPv6 connection available?

niravdesai712, 2020-02-28:
[BR_Config_json.txt](https://github.com/ARMmbed/mbed-cloud-client-example/files/4266449/BR_Config_json.txt)
This are my BR configuration any change require in this???

juhhei01, 2020-02-28:
Hi. Your configure file is OK.
Now only blocker may be that your internet service provider not give IPv6 address to border router. If you connect your laptop to same Ethernet router do you get Global IPv6 address.

Border router learn automatic IPv6 prefix from Router Advertisement message from Internet side when Prefix have a SLAAC flag enabled.

niravdesai712, 2020-02-28:
yes i am getting IPV6 address on my laptop

juhhei01, 2020-02-28:
Ipv6 must be global not fe80::. What is IPv6 prefix? Border router will learn same prefix if address is SLAAC operated. Can you check that by using for example Wireshark and capture Router Advertisment message  and check that it include proper IPv6 prefix with SLAAC or autonous flag.

Do you have Border router logs?

niravdesai712, 2020-02-28:
[BR_LOG.log](https://github.com/ARMmbed/mbed-cloud-client-example/files/4266573/BR_LOG.log)
Boarder Router LOG.

juhhei01, 2020-02-28:
Looks like that Border router not learn automatically IPv6 prefix. It is possible that Laptop is using DHCPv6 for getting address. Currently Border router need Prefix information from Router Advertisement. What is your laptop Ipv6 address prefix?
"
2K[90m[DBG ][pIP6]: Push RS to IPV6[0m
[2K[90m[DBG ][icmp]: Possible Default Router[
"
Indicate that Border router sen Router Solication and got Router Advertisment but unfortunately border router not currently not use DHCv6 address allocation.  What kind of router box you are using. Can you configure Router Advertisment daemon setup?

But main problem is that Border router is not learn IPv6 connectivity from ethernet side. It lear Default IPv6 router but not Global prefix which is mandatory for enable End to End IPv6 commutation to cloud.

niravdesai712, 2020-02-28:
My Laptop use DHCP . 
IPV6 Address: Link-local IPv6 address:	fe80::594a:1e09:6963:e9b6%13

juhhei01, 2020-02-28:
Hi.

It looks like that you not have Global IPv6 connection available. fe80:: is link local you will need a global prefix. Like 2001::: something.

teetak01, 2020-02-28:
@niravdesai712 if you want to test that with ethernet without wisun, you can define these in your configuration to switch to IPv6.

```
            "lwip.ipv4-enabled"                         : false,
            "lwip.ipv6-enabled"                         : true
```

teetak01, 2020-04-20:
I am closing this as fixed. @niravdesai712 please reopen if you still have issues.

closed 2020-04-20 by teetak01