PelionIoT/mbed-cloud-client-example #67: pal-platform.py fails to run

RonEld, 2020-04-07:
`pal-platform.py` fails to run with the following errors:
~~~
Traceback (most recent call last):
  File "pal-platform/pal-platform.py", line 1029, in <module>
    cli(sys.argv[1:])
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 829, in _call_
    return self.main(*args, **kwargs)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 1276, in invoke
    sub_ctx = cmd.make_context(
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 700, in make_context
    self.parse_args(ctx, args)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 1048, in parse_args
    value, args = param.handle_parse_result(ctx, opts, args)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 1623, in handle_parse_result
    value = self.full_process_value(ctx, value)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 1965, in full_process_value
    return Parameter.full_process_value(self, ctx, value)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 1589, in full_process_value
    value = self.process_value(ctx, value)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 1579, in process_value
    return self.type_cast_value(ctx, value)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 1568, in type_cast_value
    return _convert(value, (self.nargs != 1) + bool(self.multiple))
  File "/tmp/python-deps/lib/python3.8/site-packages/click/core.py", line 1565, in _convert
    return self.type(value, self, ctx)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/types.py", line 46, in _call_
    return self.convert(value, param, ctx)
  File "pal-platform/pal-platform.py", line 90, in convert
    return super(DynamicChoice, self).convert(value, param, ctx)
  File "/tmp/python-deps/lib/python3.8/site-packages/click/types.py", line 181, in convert
    if not self.case_sensitive:
AttributeError: 'DynamicChoice' object has no attribute 'case_sensitive'
~~~

root cause:
requirements for `click` are `>=6.5`, and on [version 7.0](https://click.palletsprojects.com/en/7.x/changelog/) the `case_sensitive` member for `click` was introduced. Since [`DynamicChoice`](https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/pal-platform/pal-platform.py#L76) doesn't call its parent `__init__` function, I believe that it doesn't recognize `case_sensitive` as its member.
suggested solutions:
1. call `super.__init__` in the child's `__init__` function , which will use the inherited attributes (see [here](https://stackoverflow.com/questions/19205916/how-to-call-base-classs-init-method-from-the-child-class)) 
2. update the [requirements.txt](https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/pal-platform/requirements.txt) to `click>=6.5,<7.0`

korjaa, 2020-04-07:
I think workaround for now is to manually install click 7.0 before any requirements from `pal-platform/requirements.txt` and `mbed-os/requirements.txt`.

This can be done with following command:
```
pip install click==7.0
```

teetak01, 2020-04-07:
This will be fixed in 4.4.0 release version.

teetak01, 2020-04-17:
Fixed in [4.4.0](https://github.com/ARMmbed/mbed-cloud-client-example/releases/tag/4.4.0) release.

closed 2020-04-17 by teetak01