PelionIoT/mbed-cloud-client-example #69: Are there plans to update with mbed os 5.15.3 or wait for 6.0.0?

dannywahlquist, 2020-05-08:


teetak01, 2020-05-11:
HI @dannywahlquist 

We update this application along side https://github.com/ARMmbed/mbed-cloud-client as integrated package. We plan to update to future patch release when we are ready to make next Device Management Client release. Thus we do not do just lib-updates in this repository outside of normal release cycles.

Now, Mbed OS 5.15.2 introduced backwards compatibility breakage for PSA APIs (and a bug in that implementation too), thus the next version we can use will be Mbed OS 5.15.4. We also need to fix the PSA API compatiblity for the client library at the same time.

Otherwise there is no issue with current patch versions. https://github.com/ARMmbed/mbed-cloud-client is already using Mbed OS 5.15.3 (as there is no actual PSA targets in use).

teetak01, 2020-05-14:
Closing as resolved.

closed 2020-05-14 by teetak01