PelionIoT/mbed-cloud-client-example #72: Can't connect Pelion Device Management with a ATECC608A Secure element

Dimeiza, 2020-07-05:
Hi,

I have tried connecting to Pelion Device Management with a ATECC608A Secure element following this documentation.

https://www.pelion.com/docs/device-management/current/connecting/connecting-to-device-management-with-a-pre-provisioned-secure-element.html

I followed procedures of this document with compliant prerequisites, but my board(K64F) showed several error log and can't connect Pelion Device Management. 

I got a serial log by enablling mbed-trace and I noticed a difference of logs between first try and second try(with rebuild).

And while printf debugging(I'm not familier with mbed OS), I guess cause of this problem might be a difference the template_id(value of DM320118 that I used and mbed-cloud-client-example expected value).

Could you teach me a procedure to solve this problem, or could you fix your artifact(code/document)?

[Additional condition]

mbed-cloud-client-example: 4.5.0(4f758469d8d06ffb8b93ca17e6e5809aa36f2e1c)
Build envioronment: mbed-cli(1.10.2) and Mbed Studio(1.0.0) in ubuntu 18.04
DM320118 SN: 18000001654

[Serial log with first try]

------------------
Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

lfs error:493: Corrupted dir pair at 0 1
lfs error:2222: Invalid superblock at 0 1
[INFO][FSST]: KV Dir: kvstore, doesnt exist - creating new.. 
[DBG ][mClt]: M2MInterfaceFactory::create_security
[DBG ][mClt]: M2MInterfaceFactory::create_interface - IN
[INFO][mClt]: M2MInterfaceFactory::create_interface - parameters endpoint name : endpoint
[INFO][mClt]: M2MInterfaceFactory::create_interface - parameters endpoint type : default
[INFO][mClt]: M2MInterfaceFactory::create_interface - parameters life time(in secs): 86400
[INFO][mClt]: M2MInterfaceFactory::create_interface - parameters Listen Port : 0
[INFO][mClt]: M2MInterfaceFactory::create_interface - parameters Binding Mode : 9
[INFO][mClt]: M2MInterfaceFactory::create_interface - parameters NetworkStack : 1
[DBG ][mClt]: M2MInterfaceFactory::create_interface - Creating M2MInterfaceImpl
[DBG ][mClt]: M2MNsdlInterface::M2MNsdlInterface()
[DBG ][mClt]: M2MConnectionHandlerPimpl::eventloop_event_handler 0
[DBG ][mClt]: M2MObject::create_object_instance - id: 0
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MBase::M2Mbase resource name is EMPTY ===========
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 0)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 1)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 6)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 7)
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 8)
[DBG ][mClt]: M2MNsdlInterface::initialize()
[DBG ][mClt]: M2MNsdlInterface::add_object_to_list this=0x1fff24e8 object=0x1fff2b30
[DBG ][mClt]: M2MObject::set_observation_handler - handler: 0x0x1fff24ec
[DBG ][mClt]: M2MInterfaceImpl::M2MInterfaceImpl() -IN
[INFO][mClt]: M2MNsdlInterface::create_endpoint( name endpoint type default lifetime 86400, domain , mode 9)
[INFO][mClt]: M2MNsdlInterface::set_endpoint_lifetime_buffer - 86400
[DBG ][mClt]: M2MResourceBase::set_value()
[DBG ][mClt]: M2MResourceBase::report() - level 0
[DBG ][mClt]: M2MResourceBase::report() - combined level 0
[DBG ][mClt]: M2MResourceBase::report() - resource 1/0/1 is observable but not yet subscribed!
[DBG ][mClt]: M2MResourceBase::report() - mode = 1, is_observable = 1
[DBG ][mClt]: M2MNsdlInterface::registration_time - value (in seconds) 85500
[DBG ][mClt]: M2MNsdlInterface::set_retransmission_parameters() - total resend time 112
[INFO][mClt]: M2MNsdlInterface::set_retransmission_parameters() - setting max resend count to 3 with total time: 112
[DBG ][mClt]: M2MResourceBase::set_value()
[DBG ][mClt]: M2MResourceBase::report() - level 0
[DBG ][mClt]: M2MResourceBase::report() - combined level 0
[DBG ][mClt]: M2MResourceBase::report() - resource 1/0/7 is observable but not yet subscribed!
[DBG ][mClt]: M2MResourceBase::report() - mode = 1, is_observable = 1
[... 245 lines cut ...]
[DBG ][mClt]: M2MInterfaceFactory::create_interface - OUT
[DBG ][mClt]: CloudClientStorage::initialize_storage
[INFO][fcc ]: key_config_manager.c:34:kcm_init:===> 
[DBG ][fcc ]: storage_psa.c:965:storage_init:===> 
[DBG ][fcc ]: storage_psa.c:228:storage_build_item_name:===> kcm_item_name = FR_ON item_type = 4
[INFO][fcc ]: storage_psa.c:247:storage_build_item_name:<=== 
[DBG ][fcc ]: key_slot_allocator.c:206:get_active_entry_of_existing_item:===> 
[DBG ][fcc ]: item_name: :60:09:09:57:59:3a:24:f8:e5:ba:72:9c:d4:aa:ed:74:30:64:64:c5:75:9a:2f:e2:9e:de:60:25:fc:67:e1:f6:00:00
[DBG ][fcc ]: key_slot_allocator.c:142:get_ksa_item_entry:===> 
[DBG ][fcc ]: key_slot_allocator.c:193:get_ksa_item_entry:<=== result = 0, is_new_entry = 1
[DBG ][fcc ]: key_slot_allocator.c:214:get_active_entry_of_existing_item:<=== Failed to get_ksa_item_entry
[DBG ][fcc ]: key_slot_allocator.c:706:ksa_init:===> 
[DBG ][fcc ]: psa_driver_crypto.c:231:psa_drv_crypto_init:===> 
[DBG ][fcc ]: psa_driver_crypto.c:241:psa_drv_crypto_init:<=== 
[DBG ][fcc ]: psa_driver_ps.c:220:psa_drv_ps_init_reserved_data:===> 

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x3F55F
Error Value: 0xFFFFFFFE
Current Thread: lwip_tcpip Id: 0x20003E2C Entry: 0x40C15 StackSize: 0x4B0 StackMem: 0x1FFF1AA8 SP: 0x1FFF1ED8 
For more info, visit: https://mbed.com/s/error?error=0x8003010D&tgt=K64F
-- MbedOS Error Info --
Could not read PSA storage version data

jenia81, 2020-07-06:
@Dimeiza is your Atmel is pre-provisioned? Have you used Microchip tools to generate key and certificate on the Atmel SE?

Dimeiza, 2020-07-07:
@jenia81 I'm using DM320118 that I thought already pre-provisioned when I bought.
Do I need additional pre-provisioning steps for connecting to Pelion?
(I couldn't find these steps in the above document)

jenia81, 2020-07-13:
DM320118 comes empty.
The pre-provisioning of the SE is something that is done by Microchip in their factory
When you are using their DM320118  development kit, you should use [Trust Platform Design Suite](https://microchipdeveloper.com/authentication:trust-platform)
You can find this information in our documentation in our [Secure element E2E tutorial](https://www.pelion.com/docs/device-management/current/connecting/connecting-to-device-management-with-a-pre-provisioned-secure-element.html), see Note section

Dimeiza, 2020-07-15:
Yesterday, I have succeeded connecting K64F with DM320118 to Pelion Device Management without additional pre-provisioning steps for DM320118.

# solution for me

I got a following sources from TrustFLEX configurator tool(included Trust Platform Design Suite).

* tflxtls_cust_cert_def_device.c
* tflxtls_cust_cert_def_signer.c

These source includes a certificate templetes that is used to reconstruct X509 certificate from DM320118 compressed certificate.

I seem that if definition of custom credential exists, mbed-cloud-client-example uses these definitions.

So I applied these certificate definition as custom credential definitions by overwriting following files.

* source/platform/mbed-os/secure_element/se_atmel_credentials/cust_def_1_signer.c
* source/platform/mbed-os/secure_element/se_atmel_credentials/cust_def_2_device.c

And I rebuilded mbed-cloud-client-example and run it over K64F, it was connected to Pelion with no problem.

# Certificates / Key in DM320118

During this connection, I checked data sources of certificates / public key with step execution.

Mbed-cloud-client-example read a compressed device certificate from slot 10 of DM320118, and generated a device public key from slot 0. 
It also read a compressed signer signature from slot12 and read signer public key from slot 11.

After these behaviour, mbed-cloud-client-example executed atecc608a_asymmetric_sign(secure element driver function) and TLS handshake finished successfully.

According to above, It seems that DM320118 initially stores a valid device certificate associated Microchip CA.

# Suggestion

It seems that present implementation of mbed-cloud-client-example can't interpret the device certificates in DM320118 initially stored.

I recommend:

* Add a additional step(overwrite custom credential definition before build) to the documentation.

or

* Modify the implementation about certificate template.

I didn't want to emulate a production flow steps, I wanted only to evaluate a use case with a secure element.
I guess there are developers that think same to me, so I recommend update a document for supporting usecase evaluation.

jenia81, 2020-07-21:
@Dimeiza You should be able to work with ATECC608A with your own CA certificate and this is I believe what you did.
The instructions on how to do this can be seen in my comment above.
Our example should also work with the default Microchip CA that is initially stored on DM320118 - so this is something that didn't work for you out-of-the-box, am I correct?

You mentioned in your original comment:
"might be a difference the template_id(value of DM320118 that I used and mbed-cloud-client-example expected value)."
What did you mean by that? What template_id are you referencing here?

Dimeiza, 2020-07-26:
> You should be able to work with ATECC608A with your own CA certificate and this is I believe what you did.

No I didn't.
In 7/15(when I connect successfully), I haven't create any my CA certificate, and haven't add my CA certificate to Pelion. 
I have only added Microchip CA certificate in your code(mbed-cloud-client-example/source/platform/secure_element/se_atmel_credentials/default-root-ca.crt) to Pelion.

> Our example should also work with the default Microchip CA that is initially stored on DM320118 - so this is something that didn't work for you out-of-the-box, am I correct?

This point is correct.
If the document I referenced in my first comment is right, I should success to connecting my device(K64F with DM320118) to Pelion out-of-the-box(without undocumented steps), but didn't work for me.

> You mentioned in your original comment:
> "might be a difference the template_id(value of DM320118 that I used and mbed-cloud-client-example expected value)."
> What did you mean by that? What template_id are you referencing here?

This is a variable within a function(acert_set_comp_cert), it represents template ID of compressed certificate in Microchip secure element.

http://ww1.microchip.com/downloads/en/Appnotes/20006367A.pdf

When I wrote this issue, I found this problem is caused by a error return value from mcc_decompress_device_cert_chain(in mcc_atca_credential_init.c). And I investigated a call tree about this error and it is following.

* mcc_decompress_device_cert_chain->mcc_atca_read_device_cert->atcacert_read_cert->atcacert_set_comp_cert

atcacert_set_comp_cert compares template_id from comp_cert with cert_def->template_id.
But this comparison failed(template_id from comp_cert is 3 and cert_def->template_id is 2) and atcacert_set_comp_cert returned ATCACERT_E_WRONG_CERT_DEF.

If my understanding of this code is correct, comp_cert is a compressed certificate get from DM320118, and cert_def is a certificate template used by your example (actual value is g_tngtls_cert_def_2_device in tngtls_cert_def_2_device.c).

Therefore I seemed this problem is caused by a difference of a template ID between DM320118 stored and certificate template used by your example.

This is my guess, isn't the certificate template in your example (g_tngtls_cert_def_2_device) consistent the DM320118 currently on the market?
(I bought my DM320118 at june 2020)

jenia81, 2020-07-28:
This might be the issue, we purchased our DM320118 that we tested with before june2020.
we'll take a look at this and get back to you